OnionIoT/Onion-Console #5: Kernel/Repo mismatch

mh1412, 2015-11-06:
When installing some packages with `opkg install` error will occur with kernel version mismatch.

[image]

borromeotlhs, 2015-11-10:
you can try to force it. Have you attempted that?

develroo, 2015-11-11:
That won't work because it need the tun kernel module and that is compiled specifically again the kernel in the repos.

thk4711, 2015-11-27:
Can you PLEASE update the OpenWrt Package repo to the same version as your image. It should be very easy. Everything is created when you compile OpenWrt. You just have to copy that to your web server as well.

greenbreakfast, 2017-01-06:
As of version 0.1.7, this is fixed.
Please see the docs site for details: https://docs.onion.io/omega2-docs/software-using-opkg.html#onion-package-repo

closed 2017-01-06 by greenbreakfast