OnionIoT/Onion-Console #8: Status widget shows uptime excluding days

Agathia, 2015-11-09:
Status widget shows uptime as hh:mm:ss but excludes days

greenbreakfast, 2016-03-15:
Hi, this is fixed in the newer versions of the console.
Can you please upgrade to the latest Omega firmware and check again?

Agathia, 2016-03-16:
I was on 0.06 (b266), upgraded to 0.0.7 (b281). Interestingly, when I checked on 0.0.6, it was showing uptime of 40+ days (plus hh:mm:ss). Following upgrade, it just shows hh:mm:ss but I suspect that the reflash reset uptime to 0, so it would be technically correct. Will check again tomorrow and if days are shown will close the issue. Thanks.

greenbreakfast, 2016-03-16:
Yep, upgrading the firmware will cause the Omega to reboot so the uptime reset makes sense

Agathia, 2016-03-21:
Sorry for the delay. Can confirm uptime shows days. Thanks.

closed 2016-03-21 by Agathia