OnionIoT/Onion-Console #10: changing timezone, root pass, hostname still busted

boneskull, 2015-11-12:
err in console:

```[onion-settings-general::_createEventHandler]: listener method `setupWifi` not defined
12onion-session-provider.html:107 Alive```

boneskull, 2015-11-12:
I'm not familiar with polymer, but if I had to make a WAG:

```html
<button id="wifi-config-button" type="button" class="btn btn-success btn-block" on-tap="saveSettings">Save Settings</button>
```

`id="wifi-config-button"` is confusing whatever event bindings are happening.

boneskull, 2015-11-12:
if I were to send a PR to fix this, how would I set up a dev environment?

Bokenator, 2015-11-12:
@boneskull It seems to work on my end. Can you tell me which browser you are using it on? I will try to debug it.

greenbreakfast, 2016-03-15:
@boneskull can you try upgrading to the latest Omega firmware and trying again?

boneskull, 2016-03-17:
will try.

boneskull, 2016-03-26:
I can confirm this is now working.  thanks

closed 2016-03-26 by boneskull