OnionIoT/Onion-Console #14: Problems with setting timezone on latest release (also problem with Wi-Fi AP Settings-->Security)

KitBishop, 2016-03-15:
I am unable to set the timezone via the Console.  
In case it was something really silly I had done, I have now done the following (with additional notes/information) and still have the problem:
* Made sure i was on a clean copy of latest release (0.0.7 b281) by doing **oupgrade -force**
* Did a factory reset by pressing and holding reset button for 15 seconds
* At this stage:
    * **/etc/TZ** contains **GMT0**
    * **/etc/config/system** has:
	option zonename 'UTC'
	option timezone 'GMT0'
    * **Console-->Settings-->General Settings-->Timezone** shows **Select your timezone**
* Selected a time zone and clicked on **Save Settings**
* Results:
    * No message on screen
    * **/etc/TZ** and **/etc/config/system** unchanged from above
* Exited Console and rechecked files - still no change
* Did a **reboot** and rechecked files - still no change
* The only way I can get my timezone set is via one or other of the following:
    * Hand edit the**/etc/config/system** file and reboot
    * Use appropriate **uci set** and **uci commit** commands
* I also think that there really is a bug in **/www/service/onion-device-provider/onion-device- provider.html** the body of **zonenamePromise** sets **timezone** and is exactly the same code as in **timezonePromise** - this can't be right!

In addition to this, I have a similar problem in trying to set the **Security** setting in **Console-->Settings-->Wi-Fi AP Settings** - no matter what I select, the **Security** setting is ignored and I have to hand edit the file **/etc/config/wireless** to get what I want.

greenbreakfast, 2016-03-16:
We've tested both of these on our end and both are working correctly.

I've made albums of both processes, showing how the changes take effect:
Changing the Timezone: http://imgur.com/a/KoIug
Changing the AP Security: http://imgur.com/a/jbQt4

The fact that both of these work on our end seems to point to a browser caching issue. Can you please try again in an Incognito tab, or clear your browser cache and try again in a regular tab.

KitBishop, 2016-03-16:
Really puzzled by this.
I have retried after ensuring browser cache is cleared.
Have tried with 3 different browsers (with cleared cache in each case): Firefox, Chrome and MS Edge - get exactly the same problem in each case.

Is there any possibility that after an **oupgrade** there are files from an earlier version that are not getting replaced/updated?

Do you see the issue with the code in **/www/service/onion-device-provider/onion-device- provider.html** I describe above?

However, since I have a work around to handle my issue(s) and since you are not able to reproduce it/them, it is probably not worth your while spending any more time on this - I'm sure you have plenty more important stuff :-)

My apologies for putting two issues in one report - I thought they might potentially be related since they both involved config files not getting updated as expected - I will keep issues separate in future.

Anyway, thanks for your time on this :-)

greenbreakfast, 2016-04-03:
Can you please try again with the v0.0.8 firmware? I paid special attention to the timezone promises :)

KitBishop, 2016-04-03:
Will do.  Expect to have time to upgrade to 0.0.8 sometime in the next couple of days.  Will report results

KitBishop, 2016-04-04:
Have now upgraded to v0.0.8 - and in general I like the improvements.
I have had a look at the changed timezone promises and they look much simpler and tidier :-)
However, no matter what I try, I still cannot change timezone via the console :-(

greenbreakfast, 2016-04-04:
:(
Have you tried on a different Omega?

KitBishop, 2016-04-05:
I only have one Omega :-(
It is not a major problem for me since I can set the timezone to what I want by hand editing the /etc/config/system file.
So probably not worth you spending any more time on this if others aren't seeing the problem

greenbreakfast, 2016-04-05:
ok, thanks for the debug effort!

closed 2016-04-05 by greenbreakfast

asez73, 2016-04-09:
Hi, sorry but I have exactly the same issues as described here.
I feel quit worried about this as some other issues appeared in the console and terminal which really are problems. I will open separated issues for those any way.