OnionIoT/Onion-Console #15: console doesn't grok SSID with a space in it

boneskull, 2016-03-26:
I have a network with a space in the name (`SON OF ZOLTAR`).  I'm unable to connect to wifi via the console (I had to use terminal to configure it).

When displayed in the UI, it looks like this:

[image]

greenbreakfast, 2016-04-03:
Can you please try again with the v0.0.8 firmware release? We've overhauled how the WiFi Settings work!

asez73, 2016-04-12:
Hi again @greenbreakfast ,
On the firmware 0.0.8, I can't change the time zone and the wifi settings won't update neither.
On the firmware 0.0.5 which was running before, the wifi settings could not be set through the Console/Settings but had to be set up from wifisetup on a command line in a terminal running on the usb port. This is also true on 0.0.8.
The timezone settings were correctly changed on 0.0.5 however.

greenbreakfast, 2016-04-12:
@asez73 Can you elaborate on how the wifi settings won't update? Maybe post some screenshots?

asez73, 2016-04-14:
Hi @greenbreakfast ,
So, Wifi settings won't update but it could be a disconnection / reconnection to the AP problem or even a Firefox cache problem.
I have some screenshots but they don't show any thing wrong unless you know the chronology...
So here come the screen shots:
a- use a connection to my access point, which was defined from ssh terminal using wifisetup:
[image]
b- create a new connection to another of my available access points, i had to refresh the AP list by the scan button:
[image]
c- It did worked but showed on screen only after rebooting the Omega and reconnecting to it. This is probably a Firefox cache problem. Note that the domino-00C is also connected and bridged to my Freebox AP through ethernet.

As a temporary conclusion: I think, it is simpler to reboot the Omega once some changes are done in the wifi settings and eventually wipe the Firefox cache.
A deeper investigation would mean upgrading firmware then resetting to factory conditions and take whole screenshots of the problem: It's just a bit too long to do now for me.

asez73, 2016-04-14:
Hi again, this time is about timezone:
a- it is always shown as unset, including after a reboot of the omega:
[image]

b- I can select and validate my timezone:
[image]

c- after this and a refresh of the web page, it still shows as unset: identical screenshot as in (a).

d- when I connect to the Terminal application, I can see that time and date are wrong as well as the timezone is still utc:
[image]

greenbreakfast, 2016-04-14:
Yes, we're experiencing some issues with Firefox with the current release. We plan to fix that in the next two or three weeks.

For now, please use Chrome for the optimal Console experience.