OnionIoT/Onion-Console #16: Dead keys in Terminal from Console

asez73, 2016-04-09:
I have a french keyboard and some keys are just not doing anything inside the terminal from the console.
It works correctly in the editor though.
It also works from the ssh terminal opened from Ubuntu directly to the IP of the Omega on my wifi network.

Firmware version 0.0.8 b297
Firefox browser 45.0.1

greenbreakfast, 2016-04-10:
ok, we'll look into the use of the '%' character from the Settings App

asez73, 2016-04-11:
@greenbreakfast , I think I did not explained clearly: this is not a character problem, but a problem happening on different keys on the keyboard: they are all on the right side (2 last columns) of the keyboard.
Furthermore, they all work well in the editor...
The wifi configuration in settings is also suffering from this problem.

greenbreakfast, 2016-04-12:
Can you try changing the keyboard layout on your OS?
Here are some instructions for Windows: https://support.office.com/en-us/article/Switch-between-different-languages-by-using-the-Language-bar-1c2242c0-fe15-4bc3-99bc-535de6f4f258
And for Mac: http://www.dotkam.com/2011/01/04/edit-keyboard-layout-on-mac/

Let me know how it goes.

asez73, 2016-04-12:
Hi @greenbreakfast , I am sorry but I won't do something that might break my system!!!
Never try to change your keyboard configuration, otherwise, due to hidden passwords, you will keep you from logging back again...
Now, it's very strange that such a bug could be linked to my ubuntu configuration while it happens exclusively in a Firefox/Polymer/Console/Terminal application and not in an other Firefox/Polymer/Console/Editor application for instance.

asez73, 2016-04-14:
Hi @greenbreakfast , so I tried a bit differently:

a- from the editor application, I typed my keyboard keys in order left to right, top to bottom, one line not shifted and one line shifted. This gives ther result below
[image]

b- When connected to the Terminal application, I did again typed the same keys in the same order with one difference: Each line is prefixed by a # so that the terminal doesn't do anything with it:
[image]

c- As you can see, the one or two last keys (on the right just before the return key) from each row are not taken in the terminal while they are in the editor... Explicitely the keys "°/)", "$/£", "ù/%", "*/µ", "://", "!/§" are totally ignored... This doesn't happens in a terminal launch through ssh as show below:

> BusyBox v1.23.2 (2016-04-05 00:44:03 UTC) built-in shell (ash)

   ____       _             ____
  / __ \___  (_)__  ___    / __ \__ _  ___ ___ ____ _
 / /_/ / _ \/ / _ \/ _ \  / /_/ /  ' \/ -_) _ `/ _ `/
 \____/_//_/_/\___/_//_/  \____/_/_/_/\__/\_, /\_,_/
 W H A T  W I L L  Y O U  I N V E N T ? /___/
 -----------------------------------------------------
   Ω-ware: 0.0.8 b297
 -----------------------------------------------------
root@Omega-17E8:~# #²&é"'(-è_çà)=
root@Omega-17E8:~# #³1234567890°+
root@Omega-17E8:~# #azertyuiop^$
root@Omega-17E8:~# #AZERTYUIOP¨£
root@Omega-17E8:~# #qsdfghjklmù*
root@Omega-17E8:~# #qsdfghjklm%µ
root@Omega-17E8:~# #<wxcvbn,;:!
root@Omega-17E8:~# #>WXCVBN?./§
root@Omega-17E8:~# uname
Linux
root@Omega-17E8:~# uname -a
Linux Omega-17E8 3.18.29 #2 Tue Apr 5 00:55:19 UTC 2016 mips GNU/Linux
root@Omega-17E8:~# ntp
-ash: ntp: not found
root@Omega-17E8:~# nntp
-ash: nntp: not found
root@Omega-17E8:~#

greenbreakfast, 2016-04-14:
Ok, we'll try to look into this but I should warn you: this is not high on our list of priorities since it is such an edge case and can be easily remedied by using SSH or the Serial Terminal.

asez73, 2016-04-14:
@greenbreakfast I definitly agree. It's just somehow odd to see such a difference between 2 apps using the same libraries as the Terminal and the Editor do.