OnionIoT/Onion-Console #17: Problems with setting timezone

KitBishop, 2016-04-13:
This is by way of re-opening issue: https://github.com/OnionIoT/Onion-Console/issues/14
Further testing shows that the issue is browser dependent:
Firefox - setting time zone does **NOT** work
Microsoft Edge - setting time zone does **NOT** work
Chrome - setting time zone **DOES** work

greenbreakfast, 2016-04-13:
Hmm ok, this gives us a clue to work with! We'll take a look and let you know!

asez73, 2016-04-14:
Hi again @greenbreakfast , this time is about timezone with Firefox/Ubuntu 14.04:

a- it is always shown as unset, including after a reboot of the omega:
[image]

b- I can select and validate my timezone:
[image]

c- after this and a refresh of the web page, it still shows as unset: identical screenshot as in (a).

d- when I connect to the Terminal application, I can see that time and date are wrong as well as the timezone is still utc:
[image]

asez73, 2016-04-14:
So I did found a bit more about time and time zone. Actually, time and date are setup as seen below.
But the timezone is not set as the current time should be +2 hours daylight savings included...
Referring to this [page](http://martybugs.net/wireless/openwrt/timesync.cgi), I could enforce the timezone. 
Notes:

- the sign of your timezone is reversed with what is displayed on the Console settings, which is NORMAL(??).

- the dalylight savings (summer time) are apparently not accounted correctly, I had to use -2 instead of -1 in my manual setup

>root@Omega-17E8:~# date
>Thu Apr 14 08:27:54 GMT 2016
>root@Omega-17E8:~# uptime
> 08:32:10 up  1:21,  load average: 0.00, 0.02, 0.04
>root@Omega-17E8:~# cat /etc/TZ
>GMT0
>root@Omega-17E8:~# echo GMT-2:00 > /etc/TZ
>root@Omega-17E8:~# date
>Thu Apr 14 10:43:03 GMT 2016
>root@Omega-17E8:~# uname -a
>Linux Omega-17E8 3.18.29 #2 Tue Apr 5 00:55:19 UTC 2016 mips GNU/Linux
>root@Omega-17E8:~#

KitBishop, 2016-04-14:
@asez73 In relation to your posts above:
First, as per my earlier comments, setting the timezone in the console doesn't work from Firefox.  It does work from Chrome.  This is not the first time there have been issues with Firefox.

Secondly, attempting to set the time zone by changing **/etc/TZ** doesn't always work as expected.  To set the timezone manually, you need to change **/etc/config/system**
There are two ways to go about this:

a. Hand edit the file **/etc/config/system** to set the entries **option timezone** and **option zonename** in the **config system** section to appropriate values (see later).
Then reboot your Omega

b. Use **uci set** and **uci commit** commands as follows(see later for actual values for **\<timezone\>** and **\<zonename\>**):
      **uci set system.@system[0].timezone='\<timezone\>'**
      **uci set system.@system[0].zonename='\<zonename\>'**
      **uci commit system**

The **timezone** and **zonename** values that should be used can be discovered from the file **/www/apps/onion-settings/onion-settings-general.html** in the definition of **timezones** from line 93 onwards.
For each entry, the **timezone** is given by the **tz** value and the **zonename** is given by the **name** value.
Note in particular that the **tz** values include information pertaining to the zone code, it's offset from UTC and information concerning the dates at which daylight savings changes.

I deduce from the screen shots you show above that you need:
    **zonename='Brussels, Copenhagen, Madrid, Paris'**
    **timezone='CET-1CEST,M3.5.0,M10.5.0/3'**

Hope this all helps.

greenbreakfast, 2016-04-14:
Yep, everything @KitBishop said is correct. 

We will be fixing the browser issues within the next two or three weeks. I'll make sure to ping you guys to confirm it works on your end then.

For now, please use Chrome for the best Omega Console experience

asez73, 2016-04-14:
@KitBishop , thank's for your help, it worked correctly (editor method and cut/paste in /etc/config/system) with a slight detail: The console settings displays Berlin, while /etc/config/system says Brussels... That the first zonename for the timezone: it's not really important anyway as far as time and daylight savings are correct.
[image]

[image]

So thank's again.

asez73, 2016-04-14:
@greenbreakfast , yes I think this would make life simpler. However, life without Google is not uninteresting also :smiling_imp:

KitBishop, 2016-04-15:
@asez73 Glad it worked for you :-)
Regarding what the console is saying for the zone after you make the changes:
The code used is effectively keyed off the **timezone** value.  In your case the **timezone** is:
**CET-1CEST,M3.5.0,M10.5.0/3**
In the **timezones** table in **onion-settings-general.html** there are 4 entries with this timezone, they have zonenames of:
1. **Amsterdam, Berlin, Bern, Rome, Stockholm, Vienna**
2. **Belgrade, Bratislava, Budapest, Ljubljana, Prague**
3. **Brussels, Copenhagen, Madrid, Paris**
4. **Sarajevo, Skopje, Warsaw, Zagreb**
In that order, so the one with **Amsterdam, Berlin....** gets selected.
While probably not a big issue since the time will still be set correctly, perhaps (as a low priority) the Omega guys should consider a different method of handling timezones such that each entry is uniquely identified.

Photonicsguy, 2017-01-22:
If you don't have console installed, you can access the file from here:
https://github.com/OnionIoT/Onion-Console/blob/master/www/apps/onion-settings/onion-settings-general.html