OnionIoT/Onion-Console #18: cannot connect to 1 of wifi networks

kritik, 2016-12-15:
My onion device info:
```
Onion Omega2+

Device Info
580 MHz Ralink MT-7688 MIPS
Processor

128MB 580MHz DDR2
Memory

32 MB Flash Memory
Storage
```

I cannot connect to network https://www.dropbox.com/s/bmgyp6lpwbkm3nq/Screenshot%202016-12-15%2022.53.08.png?dl=0
But can connect https://www.dropbox.com/s/e6ijktc2w5cy8c7/Screenshot%202016-12-15%2022.55.55.png?dl=0

The only difference I see that in the first case I use only n WIFI network, however it's marked in docs that n should be working https://docs.onion.io/omega2-docs/omega2.html.

senolcalisan, 2016-12-23:
@kritik 
I have faced the same problem on initial setup. Please check the router enabled 802.11b/g.  
Only 802.11n transmit couldnt be seen by Omega for the moment. (it is my hypotesesis)
[image]

kritik, 2016-12-24:
@senolcalisan yes, I connected to different backup wifi network which supports b mode (screenshot 2). After setup and update I still cannot connect to n network. As far as I remember b is slower than n and if at least 1 device is using b, then all other devices cannot use n. This is what I would like to escape.

senolcalisan, 2016-12-24:
@kritik
Same here, after updating still 11n mode is not discovered by omega. Try 11g only. Its maksimum output 54mbit.