OnionIoT/Onion-Console #19: A bunch of bugs (Omega2+)

WereCatf, 2016-12-16:
1. changing password doesn't work
2. the "Enable Wi-Fi" checkbox for WiFi is always unchecked, doesn't actually reflect the configuration
3. reboot doesn't work (just logs you out of Console)
4. shutdown doesn't work (just logs you out of Console)
5. on the left-hand menu-bar the button above the settings-one doesn't work, and neither do the two below it

nickfode, 2016-12-21:
To Number 1:
It seems like changing pw isn't working because chpasswd is missing
`change-password -user root -password test -verify test`
prompts
`line 86: chpasswd: not found`

WereCatf, 2016-12-21:
I finally got around to taking a closer look and it seems none of onion-console-terminal, onion-console-editor or onion-console-webcam are installed and that's why the icons on the left-hand side bar aren't working.

dClauzel, 2016-12-26:
Password can be change without problem via terminal, using `passwd`.

WereCatf, 2016-12-26:
@dClauzel And? This is about the web-UI, not the terminal.

dafoink, 2016-12-30:
I would say that the console needs alot of work.  I would try to install the console and it spins forever.  I also get problems where it has all the apps stating that they need to be installed.  you click on install, it spins and says that is it installed and to refresh.  you do that and try to go into the app and it then says to install (tried this with the console, editor, etc)

I then restored and started over and now I get broken image links.  Just seems that it is early for this and it isnt quite finished.

WereCatf, 2016-12-30:
@dafoink I would say the whole point behind making bug-reports and giving feedback is to help them improve it. Just twiddling with one's thumbs, sitting idle and hoping they'll notice bugs or shortcomings themselves doesn't really help anyone.

dafoink, 2016-12-30:
@WereCatf not my intention to twiddle thumbs (not sure if you were trying to chastise me for my post or not).  I was trying to find out a way to post a picture to this so that they can see what I am seeing.  Right now, the console isn't usable.  so, I am trying to submit issues as I see them.  and in the meantime work around them (or try to fix myself).  but one of the selling points of the onion is the console so that non-coders can manage the device.

WereCatf, 2016-12-30:
No, I wasn't trying to chastise or insult anyone, including you. I was just 
saying it's a good thing people complain and voice their opinion on things 
they feel are lacking.

dafoink, 2016-12-30:
sounds good.  sorry, I wasn't sure :)

the console really needs some work.  I was hoping to start using the onion2 for a project that I am currently using ESP8266 for since I have the need for a wifi hotspot as well as posting data to an MQTT server.  but, I don't see onion2 as being ready for that yet.

greenbreakfast, 2017-01-06:
can you please separate these into their own issues? makes it much easier to track on our end

closed 2017-01-06 by WereCatf