OnionIoT/Onion-Console #21: Omega2 Plus wont restart

PintsizedSix40, 2016-12-28:
Whenever I click restart, it asks me to login again, than it takes me back to the main screen, and the same thing happens.

WereCatf, 2016-12-30:
@waynebonc No, there is a bug in Console where, if you try to reboot the Omega2+, you get logged out of Console, but nothing else happens. The Omega2+ does not reboot, it doesn't shutdown, it does absolutely nothing other than Console logging you out of the session.

waynebonc, 2016-12-31:
@WereCatf @PintsizedSix40 I am sorry, however I was unable to reproduce your issue. Are you using the latest firmware? Considering you are viewing the Omega's console on a web browser, what means are you using to attempt a reboot? Try directly SSHing into your Omega via a terminal and write "reboot" as a command after logging in with the default credentials. Make sure that your Omega has fully booted first! Let me know what happens. Your Omega should successfully.

WereCatf, 2017-01-01:
@waynebonc Yes, I am using the latest firmware, namely 0.1.7 b140. "Considering you are viewing the Omega's console on a web browser, what means are you using to attempt a reboot?" -- Uh, you know, the bottom-left menu, which Logout, Restart and Shutdown, with neither Restart nor Shutdown working. That's the whole damn point. And we are talking about Console, not CLI -- do notice the capital C. The web-GUI is called Console. Of course things work from CLI, but that's not what the discussion is about.

showp1984, 2017-01-04:
I guess this issue will be connected to the "password does not change over console" issues I have been facing as well.
For me both only occur with Chrome, with Firefox the password is changing and the onion is restarting properly. (happening with both Omega2 and Omega2+)
I didn't bother to look into it since I uninstalled the console anyways to get some more space on the device, so I can't provide further information.

Chrome version with issue:
55.0.2883.87 (64-bit) Linux .deb package on Ubuntu 16.04

Firefox version working:
50.1.0 "Mozilla Firefox for Ubuntu" canonical - 1.0

tim427, 2017-01-09:
Confirmed on 0.1.7 (b139)

echoshack, 2017-01-20:
Confirm with latest Version: 0.1.9 b149