OnionIoT/Onion-Console #22: Can't Connect to Omega2 Plus with COM Port

PintsizedSix40, 2016-12-29:
Hi, I've tried to connect to my Omega2 Plus using Serial Ports, but it does not show up in device manager.
Thanks.

showp1984, 2017-01-04:
The Omega doesn't show up as a com port, the chip doing the serial interfacing does. If you don't have one and connected the Omega to your PCs internal COM port then it probably will already be installed in your device manager as COMX (where X is the port number).
Be aware however that some COM protocols send +12V or higher digital signals over the lines (depending on the implementation of the protocol) which may damage your Omega. I would strongly suggest the usage of optocouplers if connecting in such a way.
The simpler and faster solution would be to get a 3.3V TTL level serial usb interface of eBay. (< 10 bucks)

After connection simply connect with a serial console to your COM port and turn your Omega on, or if it was already turned on press Enter after connecting to trigger a visible response of your Omega.

greenbreakfast, 2017-01-06:
Hi @PintsizedSix40 
This doesn't really relate to the Console, it's more of a general Omega issue?
I'm gonna close the issue here, can you please post this on our community forum? community.onion.io

Thanks!

closed 2017-01-06 by greenbreakfast