OnionIoT/Onion-Console #23: Onion Launch Icon image src 404s

bikegriffith, 2016-12-30:
Looks like it's pointing to `/apps/onion-terminal/icon.png` instead of `/console/apps/onion-terminal/icon.png` like the sidebar is.

[image]

greenbreakfast, 2017-01-06:
can you please try using opkg to remove `onion-console-base` and then installing it again?

bikegriffith, 2017-01-06:
I would normally, but an attempted firmware update last night seems to have bricked my device.

bikegriffith, 2017-01-06:
(deleted last comment)

Looks like the latest firmware has the fix

greenbreakfast, 2017-09-17:
Yep, we had fixed it on the omega2 branch which has now been merged into master! https://github.com/OnionIoT/Onion-Console/blob/master/www/apps/onion-launcher/onion-launcher.html#L266

closed 2017-09-17 by greenbreakfast