OnionIoT/Onion-Console #26: Reboot/Shutdown not working

WereCatf, 2017-01-06:
With the bottom-left corner the menu with Restart, Shutdown and Logout menu-entries, they all just log you out. Restart or shutdown ain't working. Tested with several firmware-versions, including 0.1.7 b140

hoipolloimike, 2017-01-07:
I have also found that if I ssh in or use the console to reboot it won't reboot. It shuts down but doesn't come back on. This also happened during firmware upgrade. I had to manually switch it off and switch it back on. Firmware 0.1.7 (b139).

tim427, 2017-01-09:
Confirmed on 0.1.7 (b139)

ElanMorin, 2017-01-12:
Same on 0.1.7. (b139 & b140)

rali717, 2017-01-13:
"poweroff" does not work. It never does before really right.
But with the   0.1.8 b145  version  it makes a reboot, 
not a shutdown.

closed 2018-03-15 by WereCatf