OnionIoT/Onion-Console #27: Changing password doesn't work

WereCatf, 2017-01-06:
Console claims password was updated, but no change actually happens. Password stays the same as it was.

Tested using Firefox, Edge and Chrome under Windows 10.

torwag, 2017-01-06:
I can confirm this for a omega2+. Console claims the change but it was not put into effect. Changing the password via ssh and passwd did work

kdupreez, 2017-01-06:
Same here.. default password stays in effect.. you cant change it..

tim427, 2017-01-09:
Confirmed on 0.1.7 (b139)

algern0n, 2017-01-10:
I got the same issue on Omega2+, 0.1.7 (b139)

kdupreez, 2017-01-10:
No word from Onion??  They are very quiet and not much seems to work..  

I bought 3 x Omega2+ and they are pretty useless since the cloud doesnt work and you cant turn off AP pass through mode and you cant change default passwords.. 

meaning you have a rogue wireless access point on your network with default admin password you cant change...

WereCatf, 2017-01-10:
@kdupreez You can change the password by logging in from serial or SSH and calling `passwd`

kdupreez, 2017-01-10:
These will remain in the box until they fix the issues..  Its terrible to release something with such broken software..  they should rather have waited a couple months and released something that works.

ElanMorin, 2017-01-12:
Same on 0.1.7. (b139 & b140)

symonty, 2017-01-18:
Agree same issue just updated to lattest version, will use ssh.

geurti, 2017-01-25:
Console send : "{"code":127,"stderr":"\/usr\/sbin\/change-password: line 86: chpasswd: not found\n"}]}"

replace by passwd works : in change-password script: 
echo -e "$2\n$3" | (passwd $1)

ageeweb, 2017-02-19:
same issue
version 0.1.9 (b149)

greenbreakfast, 2017-09-17:
fixed! https://github.com/OnionIoT/Onion-Console/pull/52

closed 2017-09-17 by greenbreakfast