OnionIoT/Onion-Console #29: Onion web console settings doesnt work..

kdupreez, 2017-01-06:
- Changing Omega2+ name doesnt work.
- Changing Password doesnt work.
- Disable AP doesnt work

in general NOTHING works on this console settings page...  I cant have rogue access points with root passwords I cant change on my network..

also - the cloud device says "offline" and doesnt want to seem to connect after the firmware upgrade??

in general, not super impressed by this device's stability and unstable not working features..  

I will in all likelyhood turn it off and keep it off until the service improves..

Kinda feels like an "Alpha" release regarding stability and usablility..

krauth, 2017-01-12:
I have the same experience, the GUI doesn't change the settings! I also can't use the OLED Expansion Control.
This has nothing to do with stability of the Omega2+ though. Its stable and works very well, just the GUI doesn't...

kdupreez, 2017-01-12:
haha.. nothing works in the UI or cloud.. but "its stable and works very well".. An oxymoron if I ever heard one..   

Also not a big fan of the non standard pin headers.. you have to buy special breadboard converters for it..  

I'm hoping to change my mind when things start working and these things will be awesome, but right now, color me un-impressed..

krauth, 2017-01-12:
The console is just a little helper, you can do everything via ssh or a console. 
I just checked and the editor is actually working, which means you can even use the GUI to edit anything in your config!
The board itself is very stable and runs since I got it here... no crashes etc!

krauth, 2017-01-12:
OLED expansion works today!! Just needed a re-start!