OnionIoT/Onion-Console #30: Connecting to Cloud doesnt work..

kdupreez, 2017-01-06:
I tried registering a "new" device and it keeps on saying "offline"

Cloud service doesnt seem to work..

[image]

SrOne, 2017-01-07:
Same problem here also. I have try many times.

tim427, 2017-01-09:
Confirmed on 0.1.7 (b139)

deathlordgandhi, 2017-01-10:
Same problem here on omega2 Firmware 0.1.7 (b139)

banzy, 2017-01-10:
I've just received my Omega2+ but it's impossible to get it on the cloud.

algern0n, 2017-01-10:
Same issue, 0.1.7 (b139)

ElanMorin, 2017-01-12:
Same on 0.1.7. (b139 & b140)

AndyHPChan, 2017-01-14:
My Omega 2 and Omega 2+ with firmware version 0.1.7 (b139):
I experience the same problem that the devices kept indicating OFFLINE status even I registered the devices successfully. As a result, I can't send any cloud command to the devices through cloud service.

tchernobil, 2017-01-14:
I have the Omega2+. I updated manualy to 0.1.8 (b145) and it doesn't fix the problem. It works with my Omega1, firmware 0.1.4 (b330)

YMLkris, 2017-01-15:
Add me to the list - Omega 2, firmware updated to 0.1.7 (b139). The device's console says I'm online and give stats, but on the onion.io cloud site the device is listed as offline.

pkourany, 2017-01-18:
0.1.8b145 "fails" to register on cloud but clould.onion.io shows the device added and with the correct ID but offline all the time.

hughtonk, 2018-08-23:
I've got the same problem. Onion boards are still being sold all over the internet but is onion.io essentially a dead company now?