OnionIoT/Onion-Console #31: expled command doesn't work

hoipolloimike, 2017-01-07:
When trying to use the expled  command listed at [https://docs.onion.io/omega2-docs/expansion-dock.html#rgb-led](https://docs.onion.io/omega2-docs/expansion-dock.html#rgb-led) the console responds but nothing happens on the actual board. LED doesn't light up or change. The LED does light up red during boot, but otherwise nothing.

Also, trying to set led to straight red, blue or green results in errors (see below)

**From example on the url above:**
```
root@Omega-A633:~# expled 0xf21133
Setting LEDs to: f21133
Duty: 6 94 80
root@Omega-A633:~# 
```

**Setting to red:**
```
root@Omega-A633:~# expled 0xff0000
Setting LEDs to: ff0000
Duty: 0 100 100
Segmentation fault
Segmentation fault
```

hoipolloimike, 2017-01-10:
Just an update:
I tested the RGB LED with the GPIO commands (see below) and was able to turn on and off each of the red, green and blue LEDs. So the hardware seems good.

```
root@Omega-A633:~# gpioctl dirout-low 15
Using gpio pin 15.
root@Omega-A633:~# gpioctl dirout-low 16
Using gpio pin 16.
root@Omega-A633:~# gpioctl dirout-low 17
Using gpio pin 17.
root@Omega-A633:~# gpioctl dirout-high 15
Using gpio pin 15.
root@Omega-A633:~# gpioctl dirout-high 16
Using gpio pin 16.
root@Omega-A633:~# gpioctl dirout-high 17
Using gpio pin 17.
```

PinkySan, 2017-03-16:
with the recent firmware (0.1.10 (b160)) it seems to be working

greenbreakfast, 2017-09-17:
Upgrade to the latest firmware, this was fixed in b160: https://docs.onion.io/omega2-docs/known-firmware-issues.html#known-firmware-issues

closed 2017-09-17 by greenbreakfast