OnionIoT/Onion-Console #36: SD Card doesnt work???

kdupreez, 2017-01-12:
Popped in an SD card, but cant seem to mount nothing??

How do you make the SD card work??

davidkaste, 2017-01-17:
Upgrade to latest version of Omega firmware
`oupgrade --latest`

kdupreez, 2017-01-20:
WOOHOO!!

Its mountable and I can read/write files to SD card!

Awesome!  thanks for fixing this.

sabhay1996, 2017-01-28:
I'm having the same issue.
I'm unable to mount no options nothing coming what to do?

avygdor, 2017-01-31:
I upgraded to the latest Omega2+ firmware with oupgrade --latest -force and the problem went partially away.  I have an SD Card with an EXT4 file system.  When I insert it on a live and running system then it is recognized and works well. As shown below (via df command)
/dev/mmcblk0p1         7564456     41088   7116072   1% /tmp/run/mountd/mmcblk0p1 

However it does not show up after reboot (warm or cold).
Any ideas on how to have the SD card auto mount on reboot (i.e. be recognized)?

greenbreakfast, 2017-09-17:
[Upgrade to the latest firmware](https://docs.onion.io/omega2-docs/using-the-omega-updating.html), this was fixed in b143: https://docs.onion.io/omega2-docs/known-firmware-issues.html#known-firmware-issues

I haven't been able to reproduce the not mounting on boot. Are you still having the same issue?

closed 2017-09-17 by greenbreakfast

avygdor, 2017-09-18:
Thanks for the updated.

 

I will try the new firmware.  Right now the Onion is used by a system so I will nee to get access to another.

 

Thanks

Avygdor Moise