OnionIoT/Onion-Console #45: OLED Expansion Control does not save .lcd file to Omega2

lpeet, 2017-01-18:
When uploading an image to the OLED Expansion Control in the GUI, it is converted and displayed on the OLED, but attempting to save it to the Omega2 results in nothing.  No .lcd files are created.  Changing the save path from /root/image.lcd to something like /tmp/pic.lcd does not work either.  There also doesn't seem to be any way to do this from the command line.

I'm more interested in finding out the format of the .lcd files so I can create custom images and write them to the display in Python.  Is this documented anywhere?

Andrewwalker27, 2017-05-19:
I've noticed the same issue, tried find command on *.lcd but nothing. 
Just in case you missed it in the wiki, here's the info about the display. 
[https://docs.onion.io/omega2-docs/oled-expansion-c-library.html#programming-flow_understanding-the-display](url)
If you want to design your own graphics then there is a useful tool here and some info but the file is basically just an array of .hex characters
[https://forums.adafruit.com/viewtopic.php?f=25&t=64379](url)

hirnwunde, 2017-06-13:
Same problem here.
When i click on "Save to Omega" i get the error `TypeError: this.fileProvider.getTempFile is not a function`.

Line 253 of a2.js: `this.fileProvider.copyFile(this.fileProvider.getTempFile(), savePath);`

@lpeet 
Open your console (In Firefox `[Ctrl]+[Shift]+[K]`) and you will see the system filepath for the temporary file of the LCD-Image when you load a new image.
Something like `/tmp/tmp.icHgiL`