OnionIoT/Onion-Console #47: Can't find my Onion with http://omega-xxx.local/

danieldestro, 2017-01-21:
I first configured my Onion, connecting to its own network (Omega-XXXX) and updated the firmware.
After a reboot, I connected my laptop back to the WIFI and I could not access Onion with a web browser pointing to http://omega-xxxx.local/ (xxxx = 4 last digits of my mac address). 
I had to connect my laptop to the Onion WIFI network again to gain access to the console.
How can I fix this?

ajfisher, 2017-01-21:
If you're using chrome this generally won't work. I have this problem in chrome but in firefox and safari this will work fine. Can't vouch for IE.

An alternative option is to set a short name in the hosts file.

Other than that if you're using Linux or OSX is to just ping omega-xxxx.local from a command line and you'll get the IP address back.

PintsizedSix40, 2017-02-04:
use 192.168.3.1

danieldestro, 2017-02-04:
By the IP address I know. I expect to find it by its DNS name. Will try in a different browser. thanks

pogletree, 2017-02-08:
Works for me in Chrome... have you installed Apple's Bonjour service if using Windows?