OnionIoT/Onion-Console #48: Can't Disable--"Settings"->"Wi-Fi Access Point Setup"->"Enable(/disable) AP Mode"

sfdgsrsgrrrrrrrrrrrrr, 2017-01-21:
Options: Settings->Wi-Fi Access Point Setup->Enable AP Mode
Don't Work - Cant disable "AP mode".
Please help - how to disable AP.
When i Unchecked "Enable AP Mode" - wifi AP still working!
I don't know how to turn off Wi-Fi AP.

Model:           Onion Omega2 and Omega2+
Revision 	R1
Firmware 	0.1.9 (b149)

starwolf73, 2017-01-24:
Same here with Onion Omega2+. 
I have setup the WIFI through the Serial port with command line "wifisetup". If I try to disable the AP Mode through the web console, it clear the WIFI config and lose the connection. Have to redo the WIFI config again through the serial port. 

I try with both the firmware that came with it 0.1.5 (Not sure of the version), update to the stable version 0.1.9 (b149) but it still didn't work. I give a chance with the development version 0.1.9 (b150) but still no go. 

Definitely something with the wifisetup command line and web console that are not in sync/writing the same config. 

I didnt test yet to configure one from the AP Mode via IP 192.168.1.3 but I try later this week.

geurti, 2017-01-26:
Same problem.

```
br-wlan   Link encap:Ethernet   (Access Point)
          inet addr:192.168.3.1  Bcast:192.168.3.255  Mask:255.255.255.0
apcli0    Link encap:Ethernet  ( 
          inet addr:192.168.1.14  Bcast:192.168.1.255  Mask:255.255.255.0

```
http call is equivalent to :
`_uci set wireless.@wifi-iface[0].ApCliEnable=1_`
After reload network, that disable the wrong interface "apcli0" instead of br-wlan

jjok, 2017-03-28:
Same for me. I unticked `Enable AP mode` in `Wi-Fi Access Point Setup`, but there is still an access point.

AmTsH0csc, 2017-04-07:
same with me !
i configure the omega2+ as AP through the LAN , then every time i restart the omega or turn it off , the wifi turn on automatically and disable the LAN traffic .

later the omega had crash ;)

dastarling, 2019-07-08:
This still seems to be an issue still?  Any work around to disable AP only without disabling all WiFi?

zysoft, 2020-07-27:
The issue is still there (version 0.3.2 build 228).

I only worked with a wireless connection and can't say how will it behave with a wired one, but I have no reason to believe that there should be any difference.

The only way to disable AP that I found working is to shutdown `ra0` interface. But you have to do it **after** the client connection is made. So you boot with AP enabled, wait for the client connections to be established, and then you can shutdown the AP without side effects.

I used `ifconfig ra0 down` to turn it off. 

I didn't find any other way to do it in configuration. 

It also seems they have the `ra0` interface hardcoded deep in the driver and supporting scripts ([like this](https://github.com/OnionIoT/OpenWRT-Packages/blob/b61e82d1d8a1d6ea926eafce87f7a0909c9c7e0a/wifi-warp-core/files/lib/netifd/wireless/ralink.sh#L201)).