OnionIoT/Onion-Console #51: Reconnect to network with SSID not broadcasting.

patrikbjorkman, 2017-01-23:
Hi.

It seems like the Omega2 board doesn't reconnect to wlan where the SSID isn't broadcasted.

Order to reproduce the problem
Start the omega 2 with SSID broadcasting disable => omega 2 doesn't connect to the network.
Changing the router to broadcast SSID and then rebooting the omega 2 => omega 2 connect to the network
Changing back the router so it doesn't broadcast ssid => omega 2 still connected to the network.
Rebooting omega 2 => omega 2 doesn't connect to the network again

Have tried to upgrade to the latest firmeware but the problem is still there.

So is it possible to have the omega 2 reconnect to a wireless network where the ssid isn't broadcasted?

CryptoX, 2017-01-25:
same problem here, hidden ssid does not work, maybe there is a similar workaround like scan_ssid=1 to be added in wireless config. any hint ?

dogindot, 2017-01-27:
I'm having the same problem and can confirm the steps of reproduction of @patrikbjorkman.
I've tried the same workaround before reading this issue. Without success. It would be needful to connect to hidden networks.