OnionIoT/Onion-Console #55: Minus (-) doesn't work in Console

pmarcinkiewicz, 2017-01-30:
# Reproductions steps

1. Login into Omega and install Console.
2. Refresh Omega page and login.
3. Click Terminal icon and wait for login screen.
4. Type ls -l and Enter.

# Expected result

ls -l is send to console and list of files with all details is displayed.

# Result
        
        Login incorrect                                                                                                                            
        Omega-7851 login: root                                                                                                                     
        Password:                                                                                                                                  
                                                                                                                                                   
                                                                                                                                                   
        BusyBox v1.25.1 () built-in shell (ash)                                                                                                    
                                                                                                                                                   
           ____       _             ____                                                                                                           
          / __ \___  (_)__  ___    / __ \__ _  ___ ___ ____ _                                                                                      
         / /_/ / _ \/ / _ \/ _ \  / /_/ /  ' \/ -_) _ `/ _ `/                                                                                      
         \____/_//_/_/\___/_//_/  \____/_/_/_/\__/\_, /\_,_/                                                                                       
         W H A T  W I L L  Y O U  I N V E N T ? /___/                                                                                              
         -----------------------------------------------------                                                                                     
           Ω-ware: 0.1.9 b149                                                                                                                      
         -----------------------------------------------------                                                                                     
        root@Omega-7851:~# ls l                                                                                                                    
        ls: l: No such file or directory                                                                                                           
        root@Omega-7851:~#                                                                                                                         `
       
Firefox for OSX was used for testing:
https://www.mozilla.org/en-US/firefox/51.0.1/releasenotes/

nicolocandiani, 2017-03-01:
I have the same problem but with the simbols "<" and ">"

Andrewwalker27, 2017-05-19:
Can confirm problem still exists on Firefox and it's not just the - key affected, it also affects the ~ button on my laptop. I can also confirm problem doesn't exist on chrome.