OnionIoT/Onion-Console #56: white terminal in omega 2 plus app!!

onionam3, 2017-02-03:
hi dears 
when i open the terminal app in omega 2 plus browser its appear white window ! 
any help ?!

PintsizedSix40, 2017-02-04:
Just happened to me.
Restarting with the reset button fixed it.

onionam3, 2017-02-06:
@PintsizedSix40  i try that dear !
but not work ! 
still yet white !

ZiCog, 2017-02-15:
I get this as well.

Many times the terminal only has one black line displaying: "Omega-B5BD login:" 

Just now it repeatedly fails if I connect to it's WIFI access point address but works if I connect to it's WIFI client address on my local LAN.

I notice there is a failure logged to the browser console:

GET http://192.168.3.1/console/info.json 404 (Not Found)

at iron-request.html:297

omega2amm, 2017-02-22:
i have the same prob!!

rudihm, 2017-03-08:
The Terminal is started and is working, but the web-applet discovered the wrong IP address. It detects an internal IP-Address (in my case 192.168.179.37) instead of the bridged-wlan address i.e. 192.168.3.1 . But you could easily get to the terminal, it's accessible at the port 4200. All you have to do is, add the port number to the IP-Address of you omega i.e. http://192.168.3.1:4200 
Hope that works for you and gives the engineers the right direction to get it fixed soon.

AmTsH0csc, 2017-04-10:
wow..that great @rudihm 
its works perfectly
know ,, how can fix this issue , so no need to put this port in web in separate way ?

rudihm, 2017-04-10:
Sorry, had no time to further investigate who to fix this in the settings of the console application itself... but I'm glad that the workaround was of some value for you.