OnionIoT/Onion-Console #58: Can't install Node-red in omeage 2 plus !!

onionam3, 2017-02-06:
when i try to install Node-red in my omega 2 plus by this [link](https://wiki.onion.io/Tutorials/node-red) the install failed with reply ! 
unable to download this link for node-red !! 

any suggestions !?

kdupreez, 2017-02-20:
I was able to install standard node.js and npm..

try "opkg install node.js"  and "opkg install npm"

asandford, 2017-03-22:
node-red does run on the omega2, and here are the steps I used:
`opkg update
opkg install kmod-usb-storage-extras e2fsprogs kmod-fs-ext4
#setup for external drive#
mkdir /mnt/sda1
mkfs.ext4 /dev/sda1
mount /dev/sda1 /mnt/sda1
mkdir -p /tmp/cproot
mount --bind / /tmp/cproot
tar -C /tmp/cproot -cvf - . | tar -C /mnt/sda1 -xf -
umount /tmp/cproot
vi /etc/config/fstab    #configure like this#

config mount
    option target        /
    option device        /dev/sda1
    option fstype        ext4
    option options       rw,sync
    option enabled       1
    option enabled_fsck  0

reboot
df -h #check sda1 is mounted as /#
opkg update
opkg install nodejs npm python
npm install -g node-gyp

#Make sure all installed OK. Some of the installs may fail due to lack of memory, just wait 5 mins and retry#

mkdir -p /root/NR/tmp
mkdir -p /root/NR/cache
#this will take a very long time and thrash any media, I use a hard drive#
opkg install --cache /root/NR/cache -t /root/NR/tmp onion-node-red
#tidy up#
rm -r /usr/bin/onion-node-red/ipkg-mipsel_24kc
rm -r /usr/bin/onion-node-red/.pkgdir
cd /root/NR/tmp
rm -r *
cd /root/NR/cache
rm -r *

#run NR#
node --max-old-space-size=80 /usr/bin/onion-node-red/node_modules/node-red/red.js`
root@Omega-1521:~# uname -a
Linux Omega-1521 4.4.46 #0 Thu Feb 2 23:46:03 2017 mips GNU/Linux
root@Omega-1521:~# node --max-old-space-size=80 /usr/bin/onion-node-red/node_modules/nod
e-red/red.js

Welcome to Node-RED
===================

22 Mar 22:02:42 - [info] Node-RED version: v0.16.2
22 Mar 22:02:42 - [info] Node.js  version: v4.3.1
22 Mar 22:02:42 - [info] Linux 4.4.46 mipsel LE
22 Mar 22:02:50 - [info] Loading palette nodes
22 Mar 22:03:29 - [warn] ------------------------------------------------------
22 Mar 22:03:29 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
22 Mar 22:03:29 - [warn] [serialport] Error: Cannot find module 'serialport'
22 Mar 22:03:29 - [warn] ------------------------------------------------------
22 Mar 22:03:29 - [info] Settings file  : /root/.node-red/settings.js
22 Mar 22:03:30 - [info] User directory : /root/.node-red
22 Mar 22:03:30 - [info] Flows file     : /root/.node-red/flows_Omega-1521.json
22 Mar 22:03:30 - [info] Creating new flow file
22 Mar 22:03:30 - [info] Starting flows
22 Mar 22:03:30 - [info] Started flows
22 Mar 22:03:30 - [info] Server now running at http://192.168.3.1:1880/
`
The only problem is that serialport won't install