OnionIoT/Onion-Console #61: poweroff nor power-down -f not working

yeyeto2788, 2017-02-20:
Hello guys, 

I tried to shutdown the Omega2+ through the console button and it resets the boards, it doesn't shut it down.

I tried through the terminal in the console and also resets the board. 

Neither poweroff nor power-down -f work.

How can I successfully shut down the board?

LoyalNine1487, 2017-03-27:
As of right now the best way to turn off is.
1. poweroff
2. wait 30-60 seconds 
3. Unplug/Turn off power source.

yeyeto2788, 2017-08-28:
Thanks for that @LoyalNine1487 but it is not working for me :( might it be that I'm missing something?

yeyeto2788, 2018-01-15:
Any updates here?

SalvorinFex, 2018-01-15:
@yeyeto2788 Just a quick question, How does 
"waiting 30-60 seconds then Unplug/Turn off power source" 
not work??

SalvorinFex, 2018-01-18:
I think you are waiting for something on the Omega to happen, well you shouldn't be.
You are just supposed to unplug it after waiting the 30 to 60 seconds.
That's all!  The webpage for it won't change to indicate that it's safe to do so.
Just remove the power.  I have had this same issue for a while now.

yeyeto2788, 2018-01-20:
@SalvorinFex thanks for your response.

I'm waiting for my Omega to shutdown when it actually doesn't do it. That's the main reason I opened an issue. As you said taking the power is a solution but I guess that I have to wait for the Omega to be properly off. 

Taking the power out maybe will cause the Omega to have the storage corrupted (I think so, I'm not 100% sure about it)

As far as I know once you type the command the Omega should do a proper shutdown but it actually reboots.

yeyeto2788, 2020-01-14:
Closing issue since it seems that this won't be fixed.

Regards.

closed 2020-01-14 by yeyeto2788