OnionIoT/Onion-Console #64: Onion Console Not Loading

anubisalpha, 2017-03-07:
After having lots of issues setting it up with the upgrade failing and console not installing, it was just easier to use command-line to set it all up.

But after all that it seems ok but i am unable to access the console, it is just displaying a blank screen.

I have removed and reinstalled onion-console-base and still not working, anyone have any thoughts?

I am on FW 0.1.0.10 b160
Omega2+ with a basic expansion board.

raspberrycoulis, 2017-03-09:
I too have this problem. Updated to the latest firmware (0.1.10 b160) and needed to reset to factory settings. Followed the instructions as outlined here (https://docs.onion.io/omega2-docs/factory-reset.html) - did the command line version first - on boot I could only see the "Index of /console/" web page in the browser. Then tried the "hold reset button for 10 seconds" approach which ended up with the same result.

I attempted to follow some of the instructions here (http://community.onion.io/topic/1385/resolved-console-shows-index-of-console-only-seems-like-it-reset-completely-but-didnt-complete/3) and managed to get the console installed again (opkg install onion-console-base) but this now just shows a blank console in the browser with no options.

Omega2+ with dock

mikegustafson101, 2017-03-16:
Well that's not promising to hear. I just ran into the same issue. If I'm able to solve it I'll update.

dpsgithub, 2017-03-26:
Windows internet Explorer seems to be the culprit here.. install Google Chrome and your problems is solved.

mikegustafson101, 2017-03-26:
I was always using chrome. People are using IE that have the skills to use one of these things?

dpsgithub, 2017-03-26:
Cant be sure but Chrome did an about face for the Omega on my laptop running system 7. 
DougS

raspberrycoulis, 2017-03-29:
> Windows internet Explorer seems to be the culprit here.. install Google Chrome and your problems is solved.

Nope, I was using Chrome. I haven't used IE for years! Any suggestion or indications from Onion about maybe looking into this issue??

mikegustafson101, 2017-03-29:
I was/am using Chrome too. Had the issue, and it just went away all by itself. I didn't even restart the onion. I may have closed Chrome itself, but Im sure you've already done that.  Do you have a second onion? Can you attempt to connect to it?

raspberrycoulis, 2017-03-29:
> Do you have a second onion? Can you attempt to connect to it?

No, I don't. However, I did a factory reset on a lower firmware version and it worked fine, which leads me to think that something isn't quite right in version 0.1.10 b160. A factory reset should be just that - taking everything back to how it was when it comes out of the factory - if I have to do some manual installation that involves trawling through issues that others have faced to fix, isn't a factory reset.

jimgable, 2017-03-29:
I have the same (or similar) problem. I get an error "xxx.xxx.1.55 refused to connect." But I can see the console using ssh.

dpsgithub, 2017-03-30:
Make sure ur on the same network. Even then its nothing to write home about. No better interface than the command line, i say!

jimgable, 2017-03-30:
Well, my optimism was sort lived. The web terminal is broken again. Just "xxx.xxx.1.55 refused to connect". Setup works fine. I am using SSH for good maybe. I'm worried about the other web interfaces, but I don't have those break-out boards yet.

dpsgithub, 2017-03-30:
Definately an http:// address problem

jimgable, 2017-03-30:
I have the problem whether I use http://omega-ABCD.local or if I just use omega-ABCD.local. In fact it resolves to the latter anyway.

dpsgithub, 2017-03-30:
Should have http://192.168.2.1 or something similar

jimgable, 2017-03-30:
I tried your alternative, but no go. Still can't get the web terminal to show. To be more clear, my problem is not connecting entirely. I can see the Settings page beautifully. Just something wrong with Terminal. Meanwhile I can use SSH to control the device. I would prefer if the web Terminal worked. Partly just because I don't understand what is wrong with it.
[image]

dpsgithub, 2017-03-30:
Have you tried using Putty, much easier and more stable.

raspberrycoulis, 2017-03-31:
Without trying to sound negative here, has anybody actually investigated the firmware version, which clearly is not quite right? Surely jumping through hoops to get something working after a "factory reset" is not actually a factory reset, which is the bigger issue here. I suspect that @jimgable's issue is down to the console not working, which we all have mentioned above. Can somebody from Onion maybe look at this?

anubisalpha, 2017-03-31:
Just thought i'd let people know that i managed to fix my issue.

After quite a number of resets i decided to not bother with the web setup process and use ssh,

The best thing was for me to install the onion-console-base on command line, but it did take a bit of patience.

Thanks for all your thoughts and ideas..

Marc

jorgeg73, 2017-04-03:
Hi to all I posted my problem already in a new post so i am part of your team.
My omega was working fine until yesterday.
When i just too turn it on it connected to my home network i could ssh (putty), ftp and console without any problems.
Now my scenario cant reach it, no console, ftp or ssh get connection refuse , but if i connect to the omega wifi i can connect thru putty haven't tried ftp,  console is working but if i go to terminal i also get a connection refused, 
I have already factory reset, no fix for me. 
many reset later it work but today same problem.
The omega is getting my home network fine so i don't think thats the problem.
Havent try to ssh command line. What i did yesterday i delete the wifi connections to my home network (2 routers) in the omega i think that made the difference but today i don't know what is the problem  
Thank you

leviweather, 2017-04-03:
I am having a similar issue.  I just got my Omega 2+ with Dock. I can get webpage to show color but there is no login area.

dpsgithub, 2017-04-05:
An excerpt from the book. I totally agree. GUI is fun but bigger headache!

dpsgithub, 2017-04-05:
Wonderful guide btw! Examples are exemplary!

dpsgithub, 2017-04-07:
Im assuming that you installed the uci commands from the command line. (Onion DOCS) at docs.onion.io

leviweather, 2017-04-07:
I have installed the uci commands but it still doesn't work.

dpsgithub, 2017-04-07:
A little vague. What is "it" that doesn't work. And what is "work" expectation? Computer work is very precise!

leviweather, 2017-04-07:
The web console.  Whenever I try to access it through the address: http://omega-0517.local/, I get a multicolored screen.  I think there should be a login area but I can't find it.

leviweather, 2017-04-07:
When I tried my other computer, using Google, it connected and let me login!

small1, 2017-04-11:
This is a cookie issue. You solve it by deleting cookies with the name onion in them and go to the page again. That will solve this issue

dpsgithub, 2017-04-11:
Unsure of the cookie fix, didn't have to do that. Both of my omegas are working fine now. Patience, patience...

small1, 2017-04-11:
Your cookies probably timed out then. I just did it on my Omega and that 
solved the problem immediately. So no waiting is needed :)

hackeie, 2017-11-09:
Omega2+chrome+upgrade --> Onion Console Not Loading. 
I go back and it still does not work. the other applications work

b3kz0d, 2017-11-20:
This is a cookie issue, When I did reset factory, console didn't ask me to reinstall it, I open it with different browser and  everything is ok

DheerajSingh1107, 2018-05-22:
I am getting problem In log in . Shows colored screen but not console

dpsgithub, 2018-05-22:
Only solution I know is to install and use through Google Chrome. Worked best for me!

DheerajSingh1107, 2018-05-28:
Thank you very much I solved it. 192.168.3.1/setup-wizard 
and Boom it worked ;)