OnionIoT/Onion-Console #67: Omega2+ Console does not accept hyphen as an input, and also it freaks out when using the clear command.

BigDaddyDeadpoo1, 2017-03-15:
Just got my Omega2+ today, and I'm still pretty new to Linux.

I'm trying to see if my Omega2+ recognizes the microSD I inserted by using `fdisk -1` Linux command (or possibly another command if anyone has a better suggestion) and the console doesn't seem to want to use the hyphen. It's not a hardware issue on my end; every other key works. Also upon using the `clear` command, the console jumps now to a blank space under the `root@device-username:~#` line without actually clearing anything from the console.

frogstrangler600, 2017-03-21:
Same problem with Omega2 on power dock. 
Hyphen is not shown when typed on console terminal.
I also tried backslash \ - to cut off any special hyphen handling.
That does not work either.

LoyalNine1487, 2017-03-29:
You might need to change your keyboard layout.

frogstrangler600, 2017-03-29:
Thank you for your response.
The hyphen works fine outside the console program.
If the console requires a different keyboard map it would help if that
mapping was published. :)

nova706, 2017-03-30:
Clear is still an annoying issue