OnionIoT/Onion-Console #70: After factory reset does not work omega2

iliailia1212, 2017-03-26:
After resetting factory settings does not work omega2. On 192.168.3.1 displays information in two versions as in the screenshots. Reset via the console gives the same result.
Screenshots:
https://yadi.sk/a/ewzqV-9f3GN36A

LoyalNine1487, 2017-03-26:
Ssh into the onion if possible and flash the firmware with omega2p-v0.1.9-b159.bin for omega2+ or omega2-v0.1.9-b159.bin for omega2. Here's a guide how to flash it via ssh/console https://docs.onion.io/omega2-docs/manual-firmware-installation.html

AmTsH0csc, 2017-04-10:
the same with me !!

PRSenn, 2018-05-25:
And a year later it is still the same problem!   "After resetting factory settings does not work omega2+. 
On 192.168.3.1 displays information in two versions as in the screenshots. Reset via the console gives the same result."

greenbreakfast, 2018-05-27:
@PRSenn what firmware are you running? This is fixed in b176 and up