OnionIoT/Onion-Console #72: cant connect whit ssh

jorgeg73, 2017-04-03:
Hi to all my omega was working fine until today. i started up and try to connect via terminal ssh and i get a 

**connect to host omega-xxxx.local port 22: Connection refused.**

this is if i try connecting thru my network. 
I cant see the omega is connecting fine to my network but i cannot see it.
if i  connect to the omega's wifi i can ssh thru linux terminal with out problem but not thru console, terminal i get a connection refuse.
Thanks

mskong, 2017-04-16:
I'm having the same problem too. Apparently my putty can't connect to omega-xxxx.local as well.

jorgeg73, 2017-04-16:
i fix my problem whit this 
https://community.onion.io/topic/1178/wifi-connection-problem-with-omega2/17

now have other problem