OnionIoT/Onion-Console #74: Can't change root password

isblaze, 2017-05-18:
Can't change root password via web console.

Kylescottw, 2017-07-07:
Hello,
Simply SSH into your omega through terminal.
after you have successfully logged into your omega through the terminal (instructions on how to SSH into omega can be found here: https://docs.onion.io/omega2-docs/the-command-line.html)
you simply type: "passwd"
you will then be prompted to change the omega 2 password.
[image]

freginold, 2017-09-29:
Change password also doesn't work for me through console (Chrome 61) -- no matter what I type in it accepts it, but doesn't change.  Works fine from terminal/SSH.

Kylescottw, 2017-10-04:
Hi you have to go into the files to change the root password.
If i have time i will try and look again where the code is.