OnionIoT/Onion-Console #76: "opkg update" fails to download packages

jcoutch, 2017-05-25:
When attempting to run `opkg update`, I'm getting the following messages:

```
root@Omega-A175:~# opkg update
Downloading http://repo.onion.io/omega2/packages/core/Packages.gz
*** Failed to download the package list from http://repo.onion.io/omega2/packages/core/Packages.gz

Downloading http://repo.onion.io/omega2/packages/base/Packages.gz
*** Failed to download the package list from http://repo.onion.io/omega2/packages/base/Packages.gz

Downloading http://repo.onion.io/omega2/packages/packages/Packages.gz
*** Failed to download the package list from http://repo.onion.io/omega2/packages/packages/Packages.gz

Downloading http://repo.onion.io/omega2/packages/onion/Packages.gz
*** Failed to download the package list from http://repo.onion.io/omega2/packages/onion/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://repo.onion.io/omega2/packages/core/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://repo.onion.io/omega2/packages/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://repo.onion.io/omega2/packages/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://repo.onion.io/omega2/packages/onion/Packages.gz, wget returned 8.
```

I tried running wget directly with the Packages.gz URL:

```
root@Omega-A175:~# wget http://repo.onion.io/omega2/packages/core/Packages.gz
--2017-05-25 14:56:45--  http://repo.onion.io/omega2/packages/core/Packages.gz
Resolving repo.onion.io... 52.32.127.220, 52.26.79.192
Connecting to repo.onion.io|52.32.127.220|:80... connected.
HTTP request sent, awaiting response... 403 Forbidden
2017-05-25 14:56:45 ERROR 403: Forbidden.
```

If I go to that exact URL on my desktop system, it works fine.  It's obviously not a WiFi issue, since I'm getting an HTTP response back from repo.onion.io...but not sure why it's a 403...

jcoutch, 2017-05-25:
Found the issue.  I tried using HTTPS instead of HTTP, and the file downloaded successfully.  To fix the issue, edit `/etc/opkg/distfeeds.conf` and change all the URL's to use HTTPS:

```
src/gz omega2_core https://repo.onion.io/omega2/packages/core
src/gz omega2_base https://repo.onion.io/omega2/packages/base
src/gz omega2_packages https://repo.onion.io/omega2/packages/packages
src/gz omega2_onion https://repo.onion.io/omega2/packages/onion
```

Market1G, 2017-07-06:
So, should this close the issue?

jcoutch, 2017-07-07:
If it's been fixed in the Onion firmware...sure.

worldlove521, 2021-04-03:
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/targets/ralink/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_core
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/base/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_base
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/newifi/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_newifi
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/pear/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_pear
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_packages
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_luci
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/lafite/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_lafite
Downloading http://downloads.pangubox.com:6380/pandorabox/19.02/packages/mipsel_1004kc_dsp/mtkdrv/Packages.gz
Updated list of available packages in /var/opkg-lists/19.02_mtkdrv
Downloading /Packages.gz
*** Failed to download the package list from /Packages.gz

Collected errors:
 * opkg_download: Failed to download /Packages.gz, wget returned 1.