OnionIoT/Onion-Console #78: Terminal refuses to connect - resolved if anyone else has the problem

americankiwi, 2017-06-21:
After factory resetting my omega2, I tried opening the terminal and was met continually with 192.168.20.9 refused to connect. I was pretty sure that it was because when I factory reset, I was never prompted to install the terminal. This was when I was connecting to the omega2 via omega-ABCD.local/. The way I got the omega2 to prompt the install was by connecting via 192.168.20.9. Not sure if anyone else has encountered this, but if they have, try connecting by the IP address.

MaxDZ8, 2017-09-14:
Hello, I am looking to troubleshoot a similar problem.
I don't think it's the same though but perhaps better idea to accumulate info here.

My omega is accessible by 192.168.3.1, it opens the console (the colorful web app) by its own wifi network. I used this for about a day then I added external USB key for gcc & friends.

It basically factory-reset the thing. I had to re-register it and all. Now the terminal section doesn't load in the same way as you.

The browser is refusing to connect to 192.168.1.5 (odd!)

OFC this is a different network.

Strangely enough, if I point browser to 192.168.1.5 the colorful console loads. When I tried to connect to terminal however, it said it was not installed. Who was processing previously? IDK.

Nonetheless, after a few page refresh I was able to connect to terminal. No idea what happened but I don't have time to investigate.

MAPReiff, 2017-10-03:
I think I figured it out! So I was having the same issue (when connecting from my router's assigned IP as well as its own network). I went into Chrome's incognito mode (in order to start my browser fresh), and connected via its own network. When I went to the terminal app, it prompted me to install it! After installing it worked from its local IP, and normal use is fine.

andreas-lundgren-storytel, 2018-01-02:
The issue seams to be the caching in the browser and the fact that the console is not installed after the firmware update. Here is an other solution for Chrome:

Press Ctrl + Shift + I to open the inspector. Don't mind the actual inspector, but you can now right click the reload button next to the address field in chrome and select "Empty Cache and Hard Reload". You will now be prompted to install the console again.