OnionIoT/Onion-Console #88: Can't install webcam app

fabiopjve, 2017-08-30:
I am trying to test a USB webcam, but when I try to install the webcam app, I get the login screen after a couple seconds, I login and by clicking on the webcam icon I see the installation continues. Then it asks me to refresh the browser and after that, when I click the webcam icon it asks me to install the app again.
I tried a lot of times, both on Firefox and Chromium with the same results.
I also tried to install using command line:
root@Omega-0347:~# opkg install onion-console-webcam                                                                                                                                     
Installing onion-console-webcam (0.3-1) to root...                                                                                                                                       
Downloading http://repo.onion.io/omega2/packages/onion/onion-console-webcam_0.3-1_mipsel_24kc.ipk                                                                                        
Collected errors:                                                                                                                                                                        
 * check_data_file_clashes: Package onion-console-webcam wants to install file /www/console/apps/onion-webcam/app.json                                                                   
        But that file is already provided by package  * onion-console-base                                                                                                               
 * check_data_file_clashes: Package onion-console-webcam wants to install file /www/console/apps/onion-webcam/icon.png                                                                   
        But that file is already provided by package  * onion-console-base                                                                                                               
 * check_data_file_clashes: Package onion-console-webcam wants to install file /www/console/apps/onion-webcam/index.html                                                                 
        But that file is already provided by package  * onion-console-base                                                                                                               
 * check_data_file_clashes: Package onion-console-webcam wants to install file /www/console/apps/onion-webcam/not_available.jpg                                                          
        But that file is already provided by package  * onion-console-base                                                                                                               
 * opkg_install_cmd: Cannot install package onion-console-webcam.

need4help, 2017-09-01:
exactly the same behavior here.
Using Onion omega2+ with Power Dock.

juancamax, 2017-09-18:
I have the same problem!!!
:-(

mohir, 2017-10-02:
I had the same problem,
i installed using opkg with the --force-reinstall option,
installation was fine,
but i still don't manage to get any feedback from the webcam :(

RobertElseco, 2017-11-06:
Same here, no webcam video...

But using mjpg_streamer does show video.
(mjpg_streamer -i "input_uvc.so -d /dev/video0 -y" -o "output_http.so")

Then connect to 192.168.x.x:8080:/?action=stream and watch video

I think it's something in the settings of the Webcam App..

Any help please?

DrUlysses, 2018-06-11:
Same problem