OnionIoT/Onion-Console #100: Unable to register device to Onion Cloud

corndog2000, 2018-05-19:
When I try to register device I am sent to the login page. But when I enter my credentials and click register. The page just refreshes to the login page again, not registering the device.

ScrumpyJack, 2018-06-12:
same issue

andreas-lundgren, 2018-06-24:
I had one problem when I gave the device a name with forbidden characters, space perhaps. No feedback, just silently failing. Another name ian it succeeded.

Passiday, 2018-09-15:
same here (Chrome). "Solved" by opening the address from another browser (Firefox), and it let me advance to the next step "register device". But after entering the device info and clicking "Create Device", it just refreshed the "add device" form. Now I checked the device again from Chrome and saw that it was now registered in the cloud. So there are some caching issues to blame, apparently.