OnionIoT/Onion-Console #106: The Terminal app will not install.

Jaystings, 2018-08-02:
I am having a name resolution problem which is preventing me from accessing the Onion Omega2+ from ssh. I can access it from the Console in the web browser, but the Terminal application will not install, so I can not access the CLI at all right now. Has anyone had a similar issue? I am using Ubuntu 16.

dam2k, 2018-08-04:
You must use Chrome instead of Firefox. It should work then.

rruggiero, 2018-10-13:
I am having the same problem, however I purchased 2 Omega2+ and with much frustration I got the first one to install setup and eventually install the terminal app and that app has a nuber of display issues.  But when I went through a new install on my other Omega2+ the installation was much faster (because I avoided trying to register on the cloud) but the terminal will not install on this Omega2+ at all.  I end up with a blank white screen with safari and a grey screen with a sad terminal with the message "192.168.3.1 refused to connect." 

The firmware is the latest according to the setup program even though the versions look strange.

Resetting to factory defaults achieved nothing as I still have the exact same problem with the Omega2+